supermileage/dynamometer-pcb #17: Fix potentiometer footprint

cosparks, 2023-05-20:
The through-holes on the right side of footprint are too far apart vertically.  It seems like they should be within the bounds of, and not centred at the corners of the courtyard.
- holes probably need to be brought in ~1mm towards the center of the courtyard

The potentiometer's mounting legs can be bent so that they fit the current design, so this is not a deal-breaker

KevinZhiyuanDing, 2023-10-28:
https://github.com/supermileage/dynamometer-pcb/commit/03d35e74a3f1db1f3750efc926a0eced5b3e5ced

cosparks, 2023-10-28:
@KevinZhiyuanDing I still get notifications for this repo so saw that you are working on this.

I forgot to add that the potentiometer footprint also should be rotated 90° counter-clockwise, so the nob sits in the correct position when it's zeroed.  I'm not sure if knobs for d-shaft pots are standardized, but all the ones that I checked out at lee's electronics were the same.

this is zero (closed)
[image]
but this should be zero:
[image]

And to fully open the pot you rotate clockwise 270°

closed 2023-11-11 by valentino-jaber