supermileage/dynamometer-pcb #19: More space for potentiometer

cosparks, 2023-05-21:
With everything mounted on the board, it is apparent that the spacing of the potentiometer and brake button isn't optimal.
- sensor caps and resistors can be moved to bottom of board to bottom of board
- bottom-right silkscreen label can be moved and pico can farther down in the bottom right corner

KevinZhiyuanDing, 2023-11-18:
@cosparks I moved the potentiometer down and the rotary encoder to the right slightly to make it less cramped. Could you clarify how the Pico can be moved further down because there is a mounting point in the way. Here are the changes I made so far: 
[image]
I also increased the text label size on the silk screen. 
[image]

cosparks, 2023-11-18:
looks awesome!  You could move the pico down a tiny bit without moving that mounting hole, or you could move the mounting hole so it sits to the left of the pico and move the pico all the way down so it's almost flush with the corner.  The second option would require a small redesign to the case.

Aesthetically, neither option is great, as the buttons and rotary encoder will no longer be aligned with the pico, but that probably doesn't matter that much

cosparks, 2023-11-19:
If it's going to be out of alignment anyways, I think I would prefer the second option.

LaPommeCosmique, 2023-11-20:
That makes sense. Just wondering, what would you say would be an acceptable distance for the potentiometer and the brake button? Its harder for me to visualize everything mounted on the board as we don't have a potentiometer knob.

cosparks, 2023-11-20:
It kinda looks like the brake button should come down maybe a couple millimetres, and then the potentiometer should be centred between it and the pico.  Currently it's 11m between pot and brake button, but I'd add at least 4mm so there is >=15mm between the two.

[image]

cosparks, 2023-11-20:
I should really drop off this board and case so you guys have it

LaPommeCosmique, 2023-11-20:
I see what you mean now. I wonder if we can just remove the brake button all together, and replace the potentiometer with one that has a click at 0. Aesthetically there's less clutter and I don't think we lose any functionality. We can also have something like an LED that signals whether the brake is receiving power.

cosparks, 2023-11-20:
A couple thoughts on that.
1. It's nice to have a latching button for UI stuff: brake button is already being used for editing filenames in the text dialog editor (in this context, it performs the same function as the rotary encoder switch, but I've found it's a much nicer option because holding down the encoder switch to edit characters is kind of annoying).  I don't think a potentiometer switch would be as intuitive or useful
2. It might be hard to find a high-torque option with a switch (although I'm not sure whether or not you still want a high torque potentiometer on board)

closed 2024-01-21 by JamesZhZhang