supermileage/dynamometer-pcb #20: Add on/off switch to board

cosparks, 2023-07-09:
Among other things, an on/off switch would make it easier to reset the controller.

Add to back or side.  Make sure it aligns with the case.

KevinZhiyuanDing, 2023-11-18:
@cosparks Hey Colby, would it be okay if we implement the switch externally instead of on the board itself, adding an on off switch between the power source and the power input.

cosparks, 2023-11-18:
> @cosparks Hey Colby, would it be okay if we implement the switch externally instead of on the board itself, adding an on off switch between the power source and the power input.

that sounds like a very good solution!

cosparks, 2023-11-18:
Only advantage that I can think of for having it on the board is that it would be more accessible, and could be used with any power supply.

KevinZhiyuanDing, 2023-11-25:
Ok thanks. Alex, James and I are weighting the options

closed 2024-01-21 by JamesZhZhang