supermileage/dynamometer-pcb #23: Move LCD pins and PICO pins closer together

JamesZhZhang, 2023-11-26:
Sara mentioned that theres a large number of traces going from the LCD to the PICO, and cover a large distance. This is effectively splitting the front copper plane. She recommended moving the LCD down (where the buttons and rotary encoder are) and also flipping it so the SD reader can be aligned with a face. She also recommended Moving the PICO to be more central

cosparks, 2023-11-26:
Note that the SD card slot in the LCD screen needs to remain accessible through the slit that I added to the dyno controller case.  
The pico micro-usb connector also needs to be accessible so you can plug in through the side of the case (the original plan was to have an acrylic cover with button, potentiometer and rotary encoder holes cut: with this cover on, the pico wouldn't be accessible from the middle of the board).

It looks like there is a lot of room for stuff on the back copper, so you could instead add vias and run the traces along the back copper for part of their path.  Then you wouldn't have to rearrange anything.

cosparks, 2023-11-26:
> also flipping it so the SD reader can be aligned with a face.

The SD reader is aligned with the back face of the case

JamesZhZhang, 2023-11-26:
Yup we'll account for the SD card reader and PICO micro usb slot being on the edge of the board. We'll likely be making small modifications to the case and re-printing it if we go through with these changes. To centralize the PICO, we were thinking of possibly rotating it 90 degrees such that the micro usb is facing the top or bottom edge. 

The reason why Sara wants to move the LCD and PICO closer is because the traces are effectively cutting the front copper in half right now. We'd see the same issue if we moved the traces to the back copper. Ideally we want the majority of traces to be as short as possible.

closed 2024-01-21 by JamesZhZhang